imrahil/OctoPrint-PrintHistory #3: Add more stats

imrahil, 2015-04-24:
Need more stats:
- average print time
- total cost (need settings)
- average cost
- average filament length / volume

Jerware, 2015-06-29:
I'd love to see a pie chart that illustrates the amount of filament used on success vs failures. For me, that would be more useful than comparing the pure number of successes & failures.

imrahil, 2015-06-29:
There is info about filament usage only for full, successful prints. For failures I can only calculate rough estimate based on print time. But it's definitely doable

Jerware, 2015-06-29:
I wonder if that's something Gina could expose to you?... Otherwise, charting minutes of success vs minutes spent on failure could be interesting. Probably not worth estimating filament use though. Too many variables.

arthurlutz, 2016-09-26:
+1 on average print time 

ps. thanks for your plugin it's awesome! especially when starting and trying to get red parts down to a minimum ;-)

Loopbackconnector, 2016-10-17:
It would be great if it could capture the temp settings

stwf, 2017-06-04:
+1 for the temperature settings!