imrahil/OctoPrint-PrintHistory #4: csv / excel export fails

BerndJM, 2015-05-10:
Both exports gives the following error (runing on Octoprint 1.2.0-dev-834):

Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

Even if the server is doing "nothing" so the "overloaded" isn't very possible...

imrahil, 2015-05-10:
what platform do you use? Win7/Linux/RaspberryPi/OctoPi?

BerndJM, 2015-05-10:
The latest OctoPi stable image (2015-01-31-octopi-0.11.0) with Octoprint updated to the actual devel 1.2.0-dev-834

imrahil, 2015-05-10:
Rpi1 or 2 ?
I just installed my plugin on the same version of OctoPi (but on Rpi B+) and it works fine
can you provide your log file? (.octoprint/logs/octoprint.log)

BerndJM, 2015-05-10:
Rpi2
Here the relevant part from the logfile:

2015-05-10 17:57:50,962 - octoprint - ERROR - Exception on /plugin/printhistory/export/csv [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/__init__.py", line 225, in exportHistoryData
    writer.writerow([ historyDetails["fileName"], historyDetails["timestamp"], historyDetails["success"], historyDetails["printTime"], historyDetails["filamentLength"], historyDetails["filamentVolume"], ]);
KeyError: 'printTime'

BerndJM, 2015-05-10:
Ah ... I thin I got it:
there very unsuccessfull entries with no print time in the list, deleted them and the last line in the log says KeyError: 'filamentLength'
Deleted also the entries with no filament usage and voila:
the export works

imrahil, 2015-05-10:
thanks! have to fix that

BerndJM, 2015-05-10:
Think the reason are missing lines in the history.yaml by aborted prints:
1431274784925:
  fileName: 798531_-_SA_-_clip5.gcode
  note: ''
  printTime: 54.41129803657532
  success: false
  timestamp: 1431274784.925154

imrahil, 2015-05-10:
if you still have got problematic version of your history.yaml you can try with new version (0.4).

commit a49b76c8 (2015-05-10): Fixed #4. Cleaned up code
closed 2015-05-10 by imrahil