imrahil/OctoPrint-PrintHistory #5: Cancelled print reported as "success"

arekm, 2015-05-12:
I've cancelled a print in middle (using web gui "cancel" button) but history shows it as "success" job. Cancelled shouldn't be considered a success.

arekm, 2015-05-12:
Also filament usage is not counted correctly in such case:

Filename		Date	Print time	Filament usage	
kostka-pusta.gcode 
2015-05-13 00:04	00:11:23	0.52m / 1.21cm³	
kostka-pusta.gcode 
2015-05-12 23:45	00:15:28	0.52m / 1.21cm³

00:15:28 print was 100% done, 00:11:23 was cancelled by me in middle.

imrahil, 2015-05-13:
- Something definitely went wrong... there should be "PrintCancelled" event in logs... 
```
octoprint.plugins.printhistory - INFO - event: PrintCancelled
```
if there was "PrintDone" event so it's not my fault.
- for successful prints plugin always shows full usage of filament (calculated on file upload). for canceled/failed there should be empty field (there is no info about partial usage)

arekm, 2015-05-13:
````
2015-05-12 23:45:06,122 - octoprint.plugins.printhistory - INFO - event: PrintDone
2015-05-12 23:45:06,145 - octoprint.plugins.printhistory - INFO - metadata for kostka-pusta.gcode
2015-05-12 23:45:06,147 - octoprint.plugins.printhistory - INFO - filament volume: 1.20559762002, length: 518.86647
2015-05-12 23:45:06,149 - octoprint.plugins.printhistory - INFO - printTime: 928.392655849
2015-05-12 23:45:06,152 - octoprint.plugins.printhistory - INFO - success: True, timestamp: 1431467105.6
2015-05-13 00:04:04,258 - octoprint.plugins.printhistory - INFO - event: PrintCancelled
2015-05-13 00:04:04,260 - octoprint.plugins.printhistory - INFO - metadata for kostka-pusta.gcode
2015-05-13 00:04:04,262 - octoprint.plugins.printhistory - INFO - filament volume: 1.20559762002, length: 518.86647
2015-05-13 00:04:04,262 - octoprint.plugins.printhistory - INFO - printTime: 683.185829878
2015-05-13 00:04:04,263 - octoprint.plugins.printhistory - INFO - success: True, timestamp: 1431468243.99
````

imrahil, 2015-05-13:
could you provide .metadata.yaml for this gcode ?

arekm, 2015-05-13:
````
kostka-pusta.gcode:
  analysis:
    estimatedPrintTime: 788.4597727429149
    filament:
      tool0:
        length: 518.8664699999999
        volume: 1.2055976200199523
  hash: bf77da37666f2886371f0097b58b4f0e9c57aba5
  history:
  - printTime: 928.3926558494568
    printerProfile: _default
    success: true
    timestamp: 1431467105.602021
  - printTime: 683.1858298778534
    printerProfile: _default
    success: true
    timestamp: 1431468243.989295
  - printerProfile: _default
    success: false
    timestamp: 1431468244.2773
  links: []
  statistics:
    averagePrintTime:
      _default: 805.7892428636551
    lastPrintTime:
      _default: 683.1858298778534
````

commit 61960536 (2015-05-13): There is a bug in OctoPrint but this should fix #5
closed 2015-05-13 by imrahil