imrahil/OctoPrint-PrintHistory #6: Missing filament usage info in history

arekm, 2015-05-13:
Filament usage sometimes not recorded. I just got such print:
"filament-wall-holder2.gcode 2015-05-13 04:05	03:05:22	-"

Now it didn't save filament usage info and log doesn't say why:
2015-05-13 04:05:13,049 - octoprint.plugins.printhistory - INFO - event: PrintDone
2015-05-13 04:05:13,062 - octoprint.plugins.printhistory - INFO - metadata for filament-wall-holder2.gcode
2015-05-13 04:05:13,067 - octoprint.plugins.printhistory - INFO - printTime: 11122.7004569
2015-05-13 04:05:13,069 - octoprint.plugins.printhistory - INFO - success: True, timestamp: 1431482712.2

imrahil, 2015-05-13:
Could you post your .metadata.yaml from uploads folder for this particular gcode?

arekm, 2015-05-13:
filament-wall-holder2.gcode:
  analysis:
    estimatedPrintTime: 13121.38016944108
    filament:
      tool0:
        length: 16559.34844000001
        volume: 40.745500673612014
  hash: 7a1c617baea3928c9c52a2e350722ac0c57953ec
  history:
  - printTime: 11122.700456857681
    printerProfile: _default
    success: true
    timestamp: 1431482712.198739
  links: []
  statistics:
    averagePrintTime:
      _default: 11122.700456857681
    lastPrintTime:
      _default: 11122.700456857681

imrahil, 2015-05-13:
no idea how to replicate this... closing for now. Feel free to reopen when it occurs again

closed 2015-05-13 by imrahil