imrahil/OctoPrint-PrintHistory #7: Track filament spool

kareem613, 2015-05-31:
Feature Request:
If we can mark each print in the history as coming from a spool, then we can use this to track spool inventory. 
Being able to enter what spools I have, how much they cost and how much filament is on the spool. Then for each print in the history, select a spool it came from. 
A spool inventory screen could show all the spools and how much filament remains in each.

bolsoncerrado, 2015-06-18:
kudos on that! :dancer:

kontakt, 2017-01-01:
I like that you implemented this, but is the quantity used for that spool tracked anywhere? I don't see a place for it.

Elfe, 2017-03-19:
Would it be possible to add up jobs based on tags in comments? #abs #myredabs01 #red 

You have x jobs with #myredabs01 using x m / xx cm².
...

So no need to manage spools in the first iteration.

Mavyre, 2017-08-23:
Hello,
A plugin (FilamentManager) just came out. I opened the issue #60 about implementing its API in PrintHistory.