imrahil/OctoPrint-PrintHistory #10: Print History Plugin (0.4) stopped working

tonyphoang, 2015-06-14:
I originally installed Print History Plugin (0.4) from the Plugin Manager and everything worked then one day when I went to the History tab no data showed up.

Next I went to http://plugins.octoprint.org/plugins/printhistory/

Uninstalled Print History from the Plugin Manager and rebooted
opened up terminal
```
source ~/oprint/bin/activate
cd ~/OctoPrint
git pull & git checkout devel
python setup.py clean
 python setup.py install
pip install https://github.com/imrahil/OctoPrint-PrintHistory/archive/master.zip
sudo service octoprint restart```

Tried it again but no success. History tab opens but no data shows on the tab even after I printed some things

![screen shot 2015-06-14 at 5 42 23 pm](https://cloud.githubusercontent.com/assets/12704968/8150572/c318e222-12bc-11e5-90f1-2ae896d15db6.png)

imrahil, 2015-06-15:
all history data are stored in file *history.yaml* under *uploads* folder
check this file please

tonyphoang, 2015-06-15:
pi@octopi ~/.octoprint/uploads $ cat history.yaml 
0:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: rpi_camera.gcode
  note: ''
1433799364477:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: rpi_camera.gcode
  note: ''
  printTime: 2019.9974520206451
  success: true
  timestamp: 1433799364.477017
1433800513702:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: rpi_camera2.gcode
  note: ''
  success: false
  timestamp: 1433800513.702323
1433801929993:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: rpi_camera2.gcode
  note: ''
  printTime: 1369.3485350608826
  success: true
  timestamp: 1433801929.993082
1433805467298:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: raspberri_pi_camera_case.gcode
  note: ''
  printTime: 3005.2027390003204
  success: true
  timestamp: 1433805467.298832
1433816712642:
  fileName: Feet2.gcode
  note: ''
  printTime: 9640.595685958862
  success: true
  timestamp: 1433816712.6425
1433888939778:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: Pi_Camera_Mount_Cover_no.gcode
  note: ''
  printTime: 1896.441223859787
  success: true
  timestamp: 1433888939.778049
0:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: sitting.gcode
  note: ''
1433895446141:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: sitting.gcode
  note: ''
  success: false
  timestamp: 1433895446.141886
1433896719029:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: sitting.gcode
  note: ''
  printTime: 1133.6998569965363
  success: true
[... 518 lines cut ...]
  timestamp: 1434276945.402445
1434280258415:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: Part3.gcode
  note: ''
  printTime: 2925.7522900104523
  success: true
  timestamp: 1434280258.415706
1434295375789:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: imperial_mickey.gcode
  note: ''
  printTime: 9840.334886074066
  success: true
  timestamp: 1434295375.789123
1434342292355:
  filamentLength: 25.0
  filamentVolume: 0.0
  fileName: dragon.gcode
  note: ''
  printTime: 19185.7854180336
  success: true
  timestamp: 1434342292.355178

tonyphoang, 2015-06-15:
Trying to uninstall the plugin with plugin manager gives me this error
```

Uninstalling plugin "Print History Plugin"
Not uninstalling OctoPrint-PrintHistory at
/usr/local/lib/python2.7/dist-packages, outside environment
/home/pi/oprint
Done!

```

imrahil, 2015-06-23:
Hi! 
1. Try to update to latest OctoPrint and latest version of PrintHistory plugin.
2. is your history.yaml without indentation? it clearly wrong...

closed 2015-06-27 by imrahil