imrahil/OctoPrint-PrintHistory #11: [REQUEST] Add cost per print

bolsoncerrado, 2015-06-18:
Ideally we should be able to set up an associated "price per material/spool" for each print file, but if at first stage it's too much, just set up an "average price per spool" field and calculate price per print accordingly?

imrahil, 2015-06-25:
Duplicate of #7

closed 2015-06-25 by imrahil

juliendorra, 2016-09-02:
Cost per peint should be added to history in conjunction with / from the Cost Estimator plugin http://plugins.octoprint.org/plugins/cost/ which already has a rough calculation. 
Then the calculation itself could be made better in the plugin and that would transparently make the History plugin better.

jasiek, 2016-09-03:
The cost estimator calculates the cost, prior to printing, on the frontend - using JS and a few data structures from Octoprint. I reckon you can just reimplement that in python in your code and you'll be good to go. My implementation is a bit of a hack anyway. ;)