imrahil/OctoPrint-PrintHistory #12: Cannot Export CSV or Excel

geodave810, 2015-06-26:
I received this message when attempting to export the history in either CSV or Excel format.
-----------------------------
Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.
-----------------------------------------------------------------------------------------------------------------------------------------
This happened after I seem to have shocked the printer with electrostatic charge.  Everything else with octoprint seems to be working fine.  This happened once before awhile back & just reinstalled everything to fix it.

I am running octoPrint
Version: 1.3.0-dev-2-g1796c92 (devel branch)
Print History Plugin (0.5.1)

I also have your Navbar Temperature Plugin (0.6) installed & that is working fine.

imrahil, 2015-06-26:
could you copy&paste error details from log file?

geodave810, 2015-06-26:
Is this what you are looking for?

2015-06-26 15:47:08,406 - octoprint - ERROR - Exception on /plugin/printhistory/export/csv [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/__init__.py", line 129, in exportHistoryData
    return export.exportHistoryData(self, exportType)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/export.py", line 38, in exportHistoryData
    output.append(historyDetails["printTime"] if historyDetails["printTime"] else "-")
KeyError: 'printTime'

imrahil, 2015-06-26:
Fixed in 0.6

closed 2015-06-26 by imrahil

geodave810, 2015-06-26:
Cool. That was a very quick response.  Works perfectly now.

One other thing I noticed on your instructions to upgrade to the newest version for the OctoPrint-PrintHistory.  You must have been caught by the copy & paste error with your Navbar plugin.
You have:
pip install --ignore-installed --force-reinstall --no-deps https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip

I think you meant this to be:
pip install --ignore-installed --force-reinstall --no-deps https://github.com/imrahil/OctoPrint-PrintHistory/archive/master.zip

imrahil, 2015-06-27:
Removed all this stuff - easiest way now is using Plugin Manager from Settings