imrahil/OctoPrint-PrintHistory #13: Deleting a print from history causes list to resort

pauly7300, 2015-07-12:
Regardless of the beginning sort order, when i pressing to delete an entry, the record is deleted however the list resorts into a pattern that doesn't appear obvious.  See screenshots 1 (pre-delete) and 2 (after deletion).

In addition,although likely a separate defect, if i delete a record while a print is in progress, the printer pauses on and off for ~3-5 seconds while the deletion is being processed.

I'm on a Pi2
latest Dev build of Octoprint (1.3.0 dev95...) but i have seen this behavior even prior to the 1.2 upgrade.
Running 0.6 version of print history

[image]
[image]

imrahil, 2015-07-13:
my fault - file operations should be prohibited during printing

pauly7300, 2015-07-13:
ok thanks.  That's on the deletion...  Is the sorting a different issue?

commit 424089f9 (2015-07-28): Cleanup. Fixed #13 and #14. Bumped up version to 0.7
closed 2015-07-28 by imrahil