imrahil/OctoPrint-PrintHistory #14: Total Time for all prints not displayed

geodave810, 2015-07-23:
I do not know whether this worked before, but the total time for all prints displays as a "-" only.  I looked in the octoprint.log file & this is the only error I see. 2015-07-23 18:24:26,620 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.

Here is a screen capture from my firefox browser running under Windows 7 Pro.
[image]

geodave810, 2015-07-27:
If I delete the failed prints, the total print time will show up.

imrahil, 2015-07-27:
it's known issue - I should be able to fix it probably tommorow

closed 2015-07-28 by imrahil