imrahil/OctoPrint-PrintHistory #17: Unable to read history

Akegata, 2015-09-06:
My History tab is empty, and I get this in my log:
```
2015-09-06 15:42:38,084 - octoprint - ERROR - Exception on /plugin/printhistory/history [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/__init__.py", line 64, in getHistoryData
    raise IOError("Couldn't read history data from {path}".format(path=self._history_file_path))
IOError: Couldn't read history data from /home/pi/.octoprint/data/printhistory/history.yaml
```

The history.yaml file is accessible to the pi user (and cat'ing it works fine):
```
pi@octopi ~/.octoprint/logs $ ls -lh /home/pi/.octoprint/data/printhistory/history.yaml
-rw-r--r-- 1 pi pi 21K Sep  6 15:42 /home/pi/.octoprint/data/printhistory/history.yaml
```

octoprint is run by the pi user:
```
pi@octopi ~/.octoprint/logs $ ps auxwwf | grep octoprint
pi        2304 18.0  7.6 170728 29236 ?        Sl   10:35  59:12 /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint --host=127.0.0.1 --port=5000
```

It seems like this would be a permissions issue, but I can't figure out what exactly is causing this.

imrahil, 2015-09-07:
could you provide file content from history.yaml?

Akegata, 2015-09-07:
```
1440021173752:
  fileName: close_support.gcode
  note: ''
  success: false
  timestamp: 1440021173.75264
1440021367992:
  filamentLength: 3794.7619700000037
  filamentVolume: 9.127472187915986
  fileName: close_support.gcode
  note: ''
  success: false
  timestamp: 1440021367.992722
1440021788931:
  filamentLength: 3794.7619700000037
  filamentVolume: 9.127472187915986
  fileName: close_support.gcode
  note: ''
  success: false
  timestamp: 1440021788.931184
1440022561742:
  filamentLength: 3794.7619700000037
  filamentVolume: 9.127472187915986
  fileName: close_support.gcode
  note: ''
  success: false
  timestamp: 1440022561.742791
1440023760093:
  fileName: x-carriage.gcode
  note: ''
  success: false
  timestamp: 1440023760.093949
1440024023850:
  filamentLength: 7991.208250000097
  filamentVolume: 19.221108366309625
  fileName: x-carriage.gcode
  note: ''
  success: false
  timestamp: 1440024023.850569
1440024444768:
  filamentLength: 7991.208250000097
  filamentVolume: 19.221108366309625
  fileName: x-carriage.gcode
  note: ''
  success: false
  timestamp: 1440024444.768997
1440024944401:
  filamentLength: 7991.208250000097
  filamentVolume: 19.221108366309625
  fileName: x-carriage.gcode
  note: ''
  success: false
  timestamp: 1440024944.401018
1440025436128:
  fileName: vert_Y_endstop_part.gcode
  note: ''
  success: false
  timestamp: 1440025436.128456
1440027885402:
  fileName: vert_Y_endstop_part.gcode
  note: ''
  printTime: 2429.737524986267
  success: true
  timestamp: 1440027885.402439
1440145928911:
  filamentLength: 3794.7619700000037
  filamentVolume: 9.127472187915986
  fileName: close_support.gcode
  note: ''
  success: false
[... 768 lines cut ...]
1441568141679:
  filamentLength: 1744.6942100000595
  filamentVolume: 4.196481361436638
  fileName: Endstop_X_Support_Left.gcode
  note: ''
  success: false
  timestamp: 1441568141.679014
1441570004943:
  filamentLength: 1744.6942100000595
  filamentVolume: 4.196481361436638
  fileName: Endstop_X_Support_Left.gcode
  note: ''
  printTime: 1824.8706200122833
  success: true
  timestamp: 1441570004.943727
0:
  fileName: Prusa_i3_4_x_Y_Corners.gcode
  note: ''
1441598292017:
  fileName: Prusa_i3_4_x_Y_Corners.gcode
  note: ''
  printTime: 20540.238577127457
  success: true
  timestamp: 1441598292.017892
```

imrahil, 2015-09-20:
still cannot reproduce... :-/

closed 2015-09-20 by imrahil

chancrescolex, 2015-11-01:
I have this same issue. History panel appears blank. The plugin history.yml has all the print data.

I believe this is the relevant information from octoprint.log:

    2015-11-01 19:41:56,872 - octoprint - ERROR - Exception on /plugin/printhistory/history [GET]
    Traceback (most recent call last):
      File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
        response = self.full_dispatch_request()
      File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
        rv = self.handle_user_exception(e)
      File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
        reraise(exc_type, exc_value, tb)
      File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
        rv = self.dispatch_request()
      File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
        return self.view_functions[rule.endpoint](**req.view_args)
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/__init__.py", line 64, in getHistoryData
        raise IOError("Couldn't read history data from {path}".format(path=self._history_file_path))
    IOError: Couldn't read history data from /home/pi/.octoprint/data/printhistory/history.yaml

imrahil, 2015-11-01:
if it's possible please attach your history.yaml file or send it to me (email address is in setup.py file)

chancrescolex, 2015-11-01:
history.yaml

    1443662813747:
      fileName: bearing_20151001-12423-190wjjh-0.gcode
      note: ''
      success: false
      timestamp: 1443662813.747818
    ^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@$
      filamentLength: 12200.210159998202
      filamentVolume: 29.344944373971767
      fileName: bearing_20151001-3515-2zs4hf-0.gcode
      note: ''
      success: false
      timestamp: 1443747781.243087
    1443758904471:
      fileName: Double_Spiral_Vase2_SOLID.gcode
      note: ''
      printTime: 10136.74630188942
      success: true
      timestamp: 1443758904.471105
    1443905101283:
      filamentLength: 5314.351910000132
      filamentVolume: 12.78251432864559
      fileName: Twisted_Vase_Basic.gcode
      note: ''
      printTime: 8235.745892047882
      success: true
      timestamp: 1443905101.283733
    1444275143139:
      filamentLength: 4108.9448100004465
      filamentVolume: 9.883170478532222
      fileName: Squeegee_Handle_-_Part_1.gcode
      note: ''
      printTime: 5466.479240179062
      success: true
      timestamp: 1444275143.139481
    1444277877822:
      filamentLength: 2872.753410000137
      filamentVolume: 6.909781709580335
      fileName: Squeegee_Handle_-_Part_1.gcode
      note: ''
      success: false
      timestamp: 1444277877.822802
    1444279014288:
      fileName: Squeegee_Handle_-_Part_1_(1).gcode
      note: ''
      success: false
      timestamp: 1444279014.288751
    1444280643001:
      filamentLength: 2946.488460000134
      filamentVolume: 7.087135288927386
      fileName: Squeegee_Handle_-_Part_1_(1).gcode
      note: ''
      success: false
      timestamp: 1444280643.001647
    1444283721357:
      filamentLength: 2946.488460000134
      filamentVolume: 7.087135288927386
      fileName: Squeegee_Handle_-_Part_1_(1).gcode
      note: ''
      printTime: 3036.207976102829
      success: true
      timestamp: 1444283721.357349
    1444302141971:
      filamentLength: 15565.664729996792
      filamentVolume: 37.439811253687246
      fileName: Squeegee_Handle_-_Prototype_1.gcode
      note: ''
      printTime: 16809.74215888977
      success: true
      timestamp: 1444302141.971034
    1444361865284:
      filamentLength: 15657.819169996885
      filamentVolume: 37.661468657957535
      fileName: Squeegee_Handle_-_Prototype_2.gcode
      note: ''
      success: false
      timestamp: 1444361865.284398
    1444362292848:
      fileName: Squeegee_Handle_-_Prototype_2.gcode
      note: ''
      success: false
      timestamp: 1444362292.848677
    1444362997574:
      filamentLength: 15657.819169996885
      filamentVolume: 37.661468657957535
      fileName: Squeegee_Handle_-_Prototype_2.gcode
      note: ''
      success: false
      timestamp: 1444362997.574447
    1444364860057:
      filamentLength: 16208.785240008254
      filamentVolume: 37.66146790862875
      fileName: Squeegee_Handle_-_Prototype_2.gcode
      note: ''
      success: false
      timestamp: 1444364860.057109
    1444550225827:
      filamentLength: 15331.447450000289
      filamentVolume: 35.622954316514836
      fileName: 74.5mm_Oil_Filter_Wrench.gcode
      note: ''
      printTime: 15625.571340084076
      success: true
      timestamp: 1444550225.827733
    1444591073224:
      filamentLength: 15055.867079999522
      filamentVolume: 34.982637284278894
      fileName: 74.5mm_Oil_Filter_Wrench_102.gcode
      note: ''
      printTime: 15530.26522397995
      success: true
      timestamp: 1444591073.224015
    1444807247570:
      filamentLength: 17507.414699999765
      filamentVolume: 40.67886193344127
      fileName: Squeegee_Handle_-_Prototype_2.gcode
      note: ''
      printTime: 17729.1596159935
      success: true
      timestamp: 1444807247.570324

imrahil, 2015-11-01:
you see that strange line at the beginning of the file? what the heck is that? :) if you remove it and restart Octoprint, history plugin works fine...

chancrescolex, 2015-11-01:
Thanks, that did the trick. No idea what that line was or where it came from.