imrahil/OctoPrint-PrintHistory #18: Filament usage empty

bastard, 2015-09-15:
Most of my prints do not have a value at filament usage. Yesterday I printed the same .gcode twice. Only one of them got a value in filament usage.

I would like to add the missing values by editing the file - would you please provide me the file format and a description of the fields.

imrahil, 2015-09-20:
Hi,
sometimes there is no information about filament usage provided by Octoprint's core... 
if you want add it by yourself you have to edit history.yaml file in .octoprint/data/printhistory
filament usage is described by two fields:
```
filamentLength: 3794.7619700000037
filamentVolume: 9.127472187915986
```

closed 2015-09-20 by imrahil

tedder, 2018-03-04:
@imrahil, discussing on an ancient ticket rather than opening the same issue. [the octoprint API has the estimated and 'consumed' filament info in the file interface](https://github.com/foosel/OctoPrint/blob/777c02135eeb256437081dfd81d1e0596b45ff4f/docs/api/datamodel.rst#gcode-analysis-information).