imrahil/OctoPrint-PrintHistory #20: Track User Prints

Arengorn, 2015-10-07:
It would be nice to make use of the User Access Control feature to also log which user issued which print. On a multi-user environment having something like this could be particularly useful.
@imrahil do you think this is possible?

bastard, 2015-12-01:
I agree - it should take care of the user level as well. The user should be able to edit his entry but is not allowed to delete it.

motzke, 2016-03-10:
I would like to see which user has printed what!

Pinaute, 2016-12-06:
Hi @imrahil   
What file provided data Filename Date, Print time, ... ?
I'm going to try fix the issue

danielwells, 2017-01-03:
Looks like you've already added this as an enhancement but I wanted to put in my vote for it as well.  I'd love the "user" to be added as a column in the export as well.  Thanks.

cmtec, 2017-01-05:
+1
Seems that the main feature is already implemented (in v1.1?) but I would like that the user been added  automatically. Right now I have too guess what user did the print and manually add him/her.

imrahil, 2017-01-05:
sorry but I can't - AFAIK there is no way to get current user data on backend side

imrahil, 2017-01-05:
print details are provided in event handler which is run in different thread than rest of the backend... I didn't find a way how to workaround this.

danielwells, 2017-02-09:
Has this been added as a feature request in OctoPrint core then (to get access to the user)?

WildRikku, 2024-09-02:
> Hi @imrahil Ok. Question: is "current_user.get_name()" not accessible as a plugin?

This would not be a good solution, as a print could have been started by a different user than the current user and also a print might finish when no user is logged in.

PR #103 (2024-09-03): Use 'owner' payload from PRINT_DONE which is now available. Fixes #20…
… and also fixes #96.

See https://docs.octoprint.org/en/master/events/index.html#printing - it listed "owner" for the payload so I just used that. I guess it was added in OctoPrint 1.4 and this plugin is older than that.
commit a6a1c4d3 (2024-09-03): Merge pull request #103 from WildRikku/20_96

Use 'owner' payload from PRINT_DONE which is now available. Fixes #20…
commit c6e85a7f (2024-09-03): Use 'owner' payload from PRINT_DONE which is now available. Fixes #20 and also fixes #96.
closed 2024-09-03 by guysoft