imrahil/OctoPrint-PrintHistory #24: Filament Usage blank

sargonas, 2016-01-09:
I just ran two test prints using V 0.7, and with both prints the Filament usage showed no data.

If I expand the print file to show more info in the File library however it does correctly show the filament and volume from the gcode, so OctoPrint itself *appears* to be ware of the materials data. It seems to not be recorded by the History plugin though. I've tried two prints, both were the same model, but were sliced using two different programs (Cura and Simplify3D) and exported in two different gCode flavors. (Ultimaker Native and RepRap)
[image]

technobly, 2016-01-24:
First off just wanted to say this is a super useful plugin!  Thanks for creating it :)

I have a similar issue with a Marlin based machine.  Here are my logs for a part that I first cancelled and then printed successfully.
```
2016-01-19 04:40:47,810 - octoprint.filemanager.analysis - INFO - Starting analysis of local:fish_fossilz.gcode
2016-01-19 04:45:24,864 - octoprint.filemanager.analysis - INFO - Starting analysis of local:fish_fossilz.gcode
2016-01-19 04:45:24,935 - octoprint.plugins.printhistory - INFO - event: PrintCancelled
2016-01-19 04:45:24,964 - octoprint.plugins.printhistory - INFO - metadata for fish_fossilz.gcode
2016-01-19 05:15:07,800 - octoprint.plugins.printhistory - INFO - event: PrintCancelled
2016-01-19 05:15:07,859 - octoprint.plugins.printhistory - INFO - metadata for fish_fossilz.gcode
2016-01-19 05:15:08,041 - octoprint.plugins.printhistory - INFO - filament volume: 0.0, length: 0.0
2016-01-19 05:15:08,050 - octoprint.plugins.printhistory - INFO - success: False, timestamp: 1453180507.8
2016-01-19 06:31:34,841 - octoprint.plugins.printhistory - INFO - event: PrintDone
2016-01-19 06:31:34,844 - octoprint.plugins.printhistory - INFO - metadata for fish_fossilz.gcode
2016-01-19 06:31:34,846 - octoprint.plugins.printhistory - INFO - filament volume: 0.0, length: 0.0
2016-01-19 06:31:34,847 - octoprint.plugins.printhistory - INFO - printTime: 4235.81867504
2016-01-19 06:31:34,848 - octoprint.plugins.printhistory - INFO - success: True, timestamp: 1453185090.97
```
I will try another print and grab the Terminal output, which I'm guessing is not in a format that OctoPrint-PrintHistory understands?

technobly, 2016-01-24:
Just tracing back through the code where the data comes from, it appears that maybe dual extruders are not supported yet.  I'm using Tool1 for the above logs. @sargonas is that possibly what you are doing as well?
```
# analysis - looking for info about filament usage
        if "analysis" in fileData:
            if "filament" in fileData["analysis"] and "tool0" in fileData["analysis"]["filament"]:
                filamentVolume = fileData["analysis"]["filament"]["tool0"]["volume"]    # TODO - "tool0" means there is no dual extruder support
                filamentLength = fileData["analysis"]["filament"]["tool0"]['length']
```

technobly, 2016-01-24:
Ok that does seem to be the issue.  If you are just using Tool1 like me on a dual extruder machine, this hack will work for you.  I just removed the PrintHistory plugin, and added my own copy to edit like this:

## 1. ssh into your OctoPi and create a new copy of PrintHistory
```
cd ~/.octoprint/plugins/
git clone https://github.com/imrahil/OctoPrint-PrintHistory.git
cd OctoPrint-PrintHistory/
```

---

## 2. Change `tool0` to `tool1` on lines 46 & 47
```
nano octoprint_printhistory/eventHandler.py
```
https://github.com/imrahil/OctoPrint-PrintHistory/blob/master/octoprint_printhistory/eventHandler.py#L46-L47

---

## 3. Change tab label to "History 1" to remind you it's only tool1 now (optional)
```
nano octoprint_printhistory/__init__.py
```
https://github.com/imrahil/OctoPrint-PrintHistory/blob/master/octoprint_printhistory/__init__.py#L36

---

## 4. Install the plugin in develop mode (in case you want to make edits)
```
sudo python setup.py develop
```

---

## 5. Make any further edits and restart OctoPrint after each one
```
sudo service octoprint restart
```

---
## 6. Even though length shows up now, old history blocks the total from summing
[image]

## 7. Remove all old history for a total
[image]

imrahil, 2016-01-24:
Added multi extruder support in latest version. Please check it out.

technobly, 2016-01-25:
Sweet, thanks for the update @imrahil :)

PrintHistory would not uninstall for some reason (I think it may have had to do with the fact that I installed as a manually edited plugin) I had to install it manually this way:
```
sudo pip install https://github.com/imrahil/OctoPrint-PrintHistory/archive/master.zip
```
Trying a print now... I'll report back after it's done.

technobly, 2016-01-25:
@imrahil Bummer, bad news it didn't work:
[image]

I actually don't see any logs now for PrintHistory now
```
|  Print History Plugin (0.8) = /usr/local/lib/python2.7/dist-packages/octoprint_printhistory
|  Pushbullet (0.1.5) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octobullet
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.8-py2.7.egg/octoprint/plugins/softwareupdate
|  System Command Editor (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_systemcommandeditor
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.8-py2.7.egg/octoprint/plugins/virtual_printer
|  Yamlpatcher (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_yamlpatcher
2016-01-25 15:17:25,062 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-01-25 15:17:25,074 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-01-25 15:17:25,083 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2016-01-25 15:17:25,114 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-01-25 15:17:25,747 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2016-01-25 15:17:25,770 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2016-01-25 15:17:25,784 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2016-01-25 15:17:25,799 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2016-01-25 15:17:25,847 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2016-01-25 15:17:26,044 - py.warnings - WARNING - /usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

2016-01-25 15:17:26,610 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.123
2016-01-25 15:17:26,874 - octoprint.plugins.octobullet - INFO - Connected to PushBullet
2016-01-25 15:17:27,237 - py.warnings - WARNING - /usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

2016-01-25 15:17:29,270 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 1.1
2016-01-25 15:17:29,424 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-01-25 15:17:40,793 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.123
2016-01-25 15:17:46,547 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.123
2016-01-25 15:35:45,631 - octoprint.filemanager.analysis - INFO - Starting analysis of local:flexbatter18650Px1.gcode
2016-01-25 15:35:45,712 - octoprint.plugins.autoselect - INFO - Selecting flexbatter18650Px1.gcode on local that was just uploaded
2016-01-25 15:42:23,543 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.137
2016-01-25 15:46:28,920 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.137
2016-01-25 16:02:39,719 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.137
2016-01-25 16:07:44,110 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.137
2016-01-25 16:44:19,413 - octoprint.filemanager.analysis - INFO - Starting analysis of local:flexbatter18650Px1.gcode
2016-01-25 16:44:19,956 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octobullet/__init__.py:79: DeprecationWarning: globalGet has been renamed to global_get
  snapshot_url = self._settings.globalGet(["webcam", "snapshot"])

2016-01-25 16:44:21,832 - py.warnings - WARNING - /usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

2016-01-25 16:44:22,600 - py.warnings - WARNING - /usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
```

imrahil, 2016-01-25:
logs should be in plugin_printhistory_console.log file... if it's possible you can restart octoprint in debug mode with "-d" parameter

technobly, 2016-01-25:
Ah, I see this file now.  Is that new?  Still no `filament volume: x.x, length: x.x` info though.
```
pi@octopi:~/.octoprint/logs$ cat plugin_printhistory_console.log
2016-01-25 15:17:25,849 Plugins folder: /home/pi/.octoprint/plugins
2016-01-25 15:17:25,886 Uploads folder: /home/pi/.octoprint/uploads
2016-01-25 15:17:46,504 Rendering history.yaml
2016-01-25 15:17:46,528 Returning data
2016-01-25 15:24:47,589 Rendering history.yaml
2016-01-25 15:24:47,616 Returning data
2016-01-25 15:35:56,288 Rendering history.yaml
2016-01-25 15:35:56,311 Returning data
2016-01-25 15:42:23,454 Rendering history.yaml
2016-01-25 15:42:23,483 Returning data
2016-01-25 15:42:24,901 Rendering history.yaml
2016-01-25 15:42:24,924 Returning data
2016-01-25 16:44:19,491 Handled event: PrintDone
2016-01-25 16:44:19,537 Metadata for: flexbatter18650Px1.gcode
2016-01-25 16:44:19,542 PrintTime: 4102.92801595
2016-01-25 16:44:19,607 Success: True, Timestamp: 1453740257.94
2016-01-25 16:44:20,916 Rendering history.yaml
2016-01-25 16:44:20,997 Returning data
```

And here's the history.yaml file
```
pi@octopi:~/.octoprint/data/printhistory$ cat history.yaml 
1453647768186:
  filamentLength: 104.31319999999724
  filamentVolume: 0.0
  fileName: clip175-v1.gcode
  note: ''
  printTime: 778.1197640895844
  success: true
  timestamp: 1453647768.186411
1453740257939:
  fileName: flexbatter18650Px1.gcode
  note: ''
  printTime: 4102.928015947342
  success: true
  timestamp: 1453740257.939989
```

imrahil, 2016-01-25:
another day, another new release - 0.9! :) I pretty sure that everything should be fine now. Please clear your file list (remove all gcode files and reupload them) and history before testing

technobly, 2016-01-25:
@imrahil sweet :)  Shouldn't things work without having to perform these steps?  I'm not sure all users will want to do these things.  I don't mind removing the GCODE files, but I have a bunch of them and probably won't be re-uploading them if I do.  If you'd like I can test 0.9 without removing files and see if it works first.

imrahil, 2016-01-25:
yeah, you can keep files :)

technobly, 2016-01-26:
It's working :) :boom:
[image]
"Tool0: - Tool1: 0.10m" could even be shortened to "T0: - T1: 0.10m", or even user definable "Right: - Left: 0.10m" but not a big deal at all :)

```
pi@octopi:~/.octoprint/data/printhistory$ cat history.yaml
1453647768186:
  filamentLength: 104.31319999999724
  filamentVolume: 0.0
  fileName: clip175-v1.gcode
  note: ''
  printTime: 778.1197640895844
  success: true
  timestamp: 1453647768.186411
1453740257939:
  fileName: flexbatter18650Px1.gcode
  note: ''
  printTime: 4102.928015947342
  success: true
  timestamp: 1453740257.939989
pi@octopi:~/.octoprint/data/printhistory$ cat history.yaml
1453647768186:
  filamentLength: 104.31319999999724
  filamentVolume: 0.0
  fileName: clip175-v1.gcode
  note: ''
  printTime: 778.1197640895844
  success: true
  timestamp: 1453647768.186411
1453740257939:
  fileName: flexbatter18650Px1.gcode
  note: ''
  printTime: 4102.928015947342
  success: true
  timestamp: 1453740257.939989
1453767322139:
  filamentLength: 0.0
  filamentLength2: 104.31319999999724
  filamentVolume: 0.0
  filamentVolume2: 0.0
  fileName: clip175-v1.gcode
  note: Dual extrusion
  printTime: 775.9372670650482
  success: true
  timestamp: 1453767322.13997
```

imrahil, 2016-01-26:
could you post you .metadata.yaml file from uploads folder?

technobly, 2016-01-26:
I chopped out most of the files in here because it's really long, but I think you just want to see if this is being updated with length info for both tools.  Seems like it is.  I left one old part in there (fish_fossilz) and one newer part (flexbatter18650Px1_.12ext)
```
pi@octopi:~/.octoprint/uploads$ cat .metadata.yaml
clip175-v1.gcode:
  analysis:
    estimatedPrintTime: 479.85179592588224
    filament:
      tool0:
        length: 0.0
        volume: 0.0
      tool1:
        length: 104.31319999999724
        volume: 0.0
  hash: 820c7c6c971157c0f5838441c99362cfc234d3c7
  history:
  - printTime: 778.1197640895844
    printerProfile: _default
    success: true
    timestamp: 1453647768.186411
  - printTime: 775.9372670650482
    printerProfile: _default
    success: true
    timestamp: 1453767322.13997
  links: []
  statistics:
    averagePrintTime:
      _default: 777.0285155773163
    lastPrintTime:
      _default: 775.9372670650482
fish_fossilz.gcode:
  analysis:
    estimatedPrintTime: 3178.4147531419976
    filament:
      tool0:
        length: 0.0
        volume: 0.0
      tool1:
        length: 3084.5411000002678
        volume: 0.0
  hash: fe14b9dc5e84d65645df63729c6505f863b62d16
  history:
  - printerProfile: _default
    success: false
    timestamp: 1453178724.935208
  - printerProfile: _default
    success: false
    timestamp: 1453180507.796248
  - printTime: 4235.818675041199
    printerProfile: _default
    success: true
    timestamp: 1453185090.96682
  links: []
  statistics:
    averagePrintTime:
      _default: 4235.818675041199
    lastPrintTime:
      _default: 4235.818675041199
flexbatter18650Px1.gcode:
  analysis:
    estimatedPrintTime: 3032.8514671303165
    filament:
      tool0:
        length: 0.0
        volume: 0.0
      tool1:
        length: 3332.4120000002813
        volume: 0.0
  hash: 2a0dd4e74804304f9c1a0197edd235f385bae26c
  history:
  - printTime: 4102.928015947342
    printerProfile: _default
    success: true
    timestamp: 1453740257.939989
  links: []
  statistics:
    averagePrintTime:
      _default: 4102.928015947342
    lastPrintTime:
      _default: 4102.928015947342
flexbatter18650Px1_.12ext.gcode:
  analysis:
    estimatedPrintTime: 3125.3903711989146
    filament:
      tool0:
        length: 0.0
        volume: 0.0
      tool1:
        length: 3437.3314000003334
        volume: 0.0
  hash: b1cffce65f60e4f49e6772a7e53572cfe854731c
  history:
  - printTime: 3974.0794870853424
    printerProfile: _default
    success: true
    timestamp: 1453781191.44018
  links: []
  statistics:
    averagePrintTime:
      _default: 3974.0794870853424
    lastPrintTime:
      _default: 3974.0794870853424
```

closed 2016-03-25 by imrahil