imrahil/OctoPrint-PrintHistory #27: Some prints do not show filament usage

sworah, 2016-03-25:
Just as in the title. Some prints show the filament used, some do not. There isn't any obvious reason as to why.

ericthelin, 2016-04-06:
From what I have seen (I'm just another user) if OctoPrint shows the filament usage and time estimate at the time of printing it will show in the history. But if OctoPrint doesn't know how much filament was used there is no data to put into the history. So for a real answer you would need to contact the OctoPrint team.

sargonas, 2016-06-30:
To further help folks who stumble across this issue while searching for it, I was able to 100% prevent this from happening if I simply waited to make sure Octoprint calculated the filament values *before* I hit "Print". It you wait (time will vary based on Pi and Print file size) it has *always* recorded the values for me. You just gotta be a little less trigger happy between the load and print steps. ;)

closed 2016-09-19 by imrahil