imrahil/OctoPrint-PrintHistory #28: Store Settings

ericthelin, 2016-04-06:
It would be great to be able to open up the details of each print and see what settings it was done with. This would be really useful when I want to reproduce old prints. The extreme, and likely well beyond the scope of this plugin, version would be to archive/version the entire gcode file so that one really could just hit reprint.

imrahil, 2016-09-19:
What settings do you mean? In my opinion only useful are from slicer, but they aren't always included in gcode file.
Yup, storing gcode files is well beyond the scope of this plugin.

closed 2016-09-19 by imrahil

ericthelin, 2016-09-20:
The settings that I was hoping for are fairly basic things that could be extracted from the gcode like extruder temp, bed temp, and layer size. Maybe an md5sum of the gcode so that if it is still available on the sd card it could be reprinted or identified.