imrahil/OctoPrint-PrintHistory #33: Log filament used and time for Failed Prints

gjperera, 2016-10-16:
Ideally number of failed prints is low, but there are times a print fails after 6 hours of printing. During that time your bot's been running for 6 hours and it's used a ton filament. It would be great to capture this so that users have a try value for bot usage and filament used as a whole.

Please let me know if there are other details required for this enhancement request. Thank you.

imrahil, 2016-12-18:
At the moment (ver 1.0.0) I enabled filament usage for cancelled prints. But these values are from gcode analysis fired after file upload.... so it's usage for full, successful print (probably not so useful in your case)
What can I do is to divide this filament usage by time portion used by failed print. Rough estimation but maybe will be useful....  what do you think?

AiyionPrime, 2018-01-03:
Even such an estimate would help a lot, I think.
I was just about to open a request until I saw it's already open.
Keep up the good work!

SirBastol, 2019-01-02:
agreed !  this is a must
and a notes field  so you can add a fail category and some notes .. 

ie category ( print aborted becuase of adhesion or underextrusion or layer adhesion all teh common fails should be predefined  and then add an other)  

so when you cancel a print or it gets cancelled it  asks you why..  AND all timelapses shoudl be linked to the log as well !!!  
so you can review and see what went wrong and on what layer..