imrahil/OctoPrint-PrintHistory #34: Print time blank with OctoPrint 1.3.0rc1

JohnOCFII, 2016-10-30:
Since upgrading OctoPrint to the latest release candidate, Print time is not displayed.  I've attached a screen shot.

[image]

JohnOCFII, 2016-12-11:
Since the 0.9.1 update, times are once again displayed properly.  Closing.

JohnOCFII, 2016-12-17:
I was too quick to close the issue.  It seemed better after 0.9.1, but after a few prints, it stopped displaying the time.  I just updated to 0.9.2, and haven't printed with it yet, but here is a current screen shot.  
[image]

imrahil, 2016-12-17:
two things: 
- just discovered that 0.9.2 isn't production ready yet :)
- could you share somehow your history.yaml file?

JohnOCFII, 2016-12-17:
My zipped history.yaml is attached.

[history.yaml.zip](https://github.com/imrahil/OctoPrint-PrintHistory/files/659036/history.yaml.zip)

imrahil, 2016-12-17:
Fixed in 1.0.0

closed 2016-12-17 by imrahil