imrahil/OctoPrint-PrintHistory #36: Print time and filament usage not showing

musyne, 2016-12-19:
Hi,
On a fresh Octoprint 1.3 everything is working fine except as the title says. I've done 4 prints from it so far (all successful), but I only get the Date and the status of the print, no other information ("-").
Do I need to setup something?

musyne, 2016-12-19:
New discovery: I've had a failed print and it shows the print time and filament usage. So I guess a recent commit somehow inverted the logic by mistake?

imrahil, 2016-12-20:
you have to wait for gcode analysis if you want detailed info in history tab... see: https://github.com/imrahil/OctoPrint-PrintHistory/issues/32

closed 2016-12-21 by imrahil

Tailslide, 2018-04-20:
I'm using the API to print directly from my slicer.. is there some way to insert a delay so the filament usage will show up?