imrahil/OctoPrint-PrintHistory #37: Duplicate entries and missing entries

JohnOCFII, 2016-12-22:
Greetings,

I'm using 1.0 and seeing some missing entries, double entries, and entries that are marked as failed prints.

I've attached history.db and history.bak.  For good measure, I also include my old history.yaml which ended after the upgrade, but might prove useful.  These are all zipped up in the attachment.  If I can provide any additional information, please let me know.

Thank you,

John

[PrintHistory-20161222.zip](https://github.com/imrahil/OctoPrint-PrintHistory/files/669660/PrintHistory-20161222.zip)

imrahil, 2016-12-22:
in your history.yaml (or .bak - files are identical) you have duplicates too. that's strange because only difference is in timestamp (and smaller than 1second!). These duplicates are even visible on screenshot in your previous ticket (#34)

more suspicious is  lack of info about filament volume on almost all entries...

one more thing - you cannot have history.db and history.yaml at the same time. On startup plugin will find history.yaml and will try import values into history.db (and it finish with duplicates!)

JohnOCFII, 2016-12-22:
The history.yaml I had saved off in another directory -- I just sent it along in case it was helpful to see or rule out some conversion issue when you moved to mySQL.

As far as the content of that file -- I don't know what to tell you.  I've never edited it by hand.

JohnOCFII, 2016-12-22:
I meant to attach a screen shot.  Here it is: 
[image]

lubarb, 2016-12-27:
I also had a complete print that does not appear in the history with 1.0.

I believe it was loaded by the autoselect plugin and started from the print button. It was also (I'm sure irrelevant) a spiral ornament. Probably an 8-12 hour print.

I'd be happy to try to provide more info if you can point me at some reasonably detailed directions - I've had octopi up for about a week and am running this all by cookbook.

- Luther

imrahil, 2016-12-27:
can you provide your octoprint.log and plugin_printhistory_console.log ?

lubarb, 2016-12-27:
Hi,

plugin_printhistory_console.log does not appear on the list of files on the
Settings/logs page. Octoprint.log entries are attached.

lubarb, 2016-12-27:
By the way, I really like having this capability!

imrahil, 2016-12-27:
attached files are missing...

lubarb, 2016-12-28:
I'll try again...