imrahil/OctoPrint-PrintHistory #38: Lost History with new History Upgrade

wmmc01, 2016-12-27:
Help - I completely lost all my previous History when I ran the last update to the History module. It was 10 months of printing history. Is there some way to recover that?

imrahil, 2016-12-27:
check if you have history.bak file in .octoprint/data/printhistory

wmmc01, 2016-12-27:
I don’t see that subdirectory anywhere.

imrahil, 2016-12-27:
https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#where-can-i-find-those-log-files-you-keep-talking-about

wmmc01, 2016-12-29:
Sorry, what? You told me to see if I could find a history.bak file, which I do not understand how to find it - then you tell me how to find a log file?

imrahil, 2016-12-29:
sorry. messed things... but log file could be useful too

wmmc01, 2016-12-30:
But what about the History.bak file?

imrahil, 2016-12-30:
if you'll find log files, history.bak should be there too (in data/printhistory folder)

wmmc01, 2017-01-04:
I don't see those files or even the subdirectories on my PC. Or should I be looking on the SD card with the OctoPrint image in the Raspberry Pi?

wmmc01, 2017-01-09:
Hello?

imrahil, 2017-01-09:
sorry. can't help you

closed 2017-01-09 by imrahil

wmmc01, 2017-01-21:
Seriously?