imrahil/OctoPrint-PrintHistory #42: Missing Data

danielwells, 2017-01-03:
I don't understand why sometimes there is data missing on successful prints.  Sometimes it's just the time missing.  Sometimes it's the time and the filament.  Anyone know why this might be and if I can do something to prevent it?

[image]

imrahil, 2017-01-03:
I do my best to collect all available data for prints... I don't know why sometimes users don't have some data. I have to discuss with @foosel :)

cmtec, 2017-01-05:
I have this issue too. Normally it's the filament usage that's missing for me.
Octoprint v1.3.0
Print History Plugin v1.1.1
Sliced with Simplify3D
Printed on Prusa i3 MK2 or Wanhao D4

I have seen this on earlier versions of octoprint/print history plugin as well.

sargonas, 2017-02-01:
This happened a few times for me, and it was almost always an issue of my own impatience, till I figured it out. When you upload a file to octoprint, there is a delay before it knows all of the print parameters, while it crunches some math in the background. Before that is done, the system only knows the most bare-minimum details about the print. If you start the print before the match is crunched, it actually doesn't save it anywhere, it sort of forgets it was calculating that data and goes right to printing. Then, when the plugin writes it's data, those fields in the cache are empty. If you just wait a few seconds/minute for the top-left details box to fully populate before you hit print, then you should be good to go.

TL;DR - Wait a bit before hitting print, for the fields in the "State" module to fully populate.

closed 2017-02-08 by imrahil

danielwells, 2017-02-09:
This seems like a bug in OctoPrint.  Does anyone know if it has been submitted?  it should continue to calculate this info even after the print begins.

foosel, 2017-02-09:
No, it shouldn't. Gcode analysis is an expensive process that could otherwise interfere with printing, leading to artifacts in the print or stuttering. Hence it is stopped when a print job is started and only resumed after the end of a job. This is intentional behaviour, printing always takes priority.