imrahil/OctoPrint-PrintHistory #44: Could you add Change Spool?

Vikinge, 2017-02-10:
Could you add option to track by spool or filament color?  I was tracking my black spool usage, but changed to white last night and now the data doesn't help me calculate how much of my spool has been used.  
Is there a way to reset all the data?  If I can't separate by spool or color?

I would want the print statistics to maintain across spools.

Vikinge, 2017-02-10:
Pretty much same as request #7

imrahil, 2017-02-10:
it's already implemented. Spools you can add in settings, then set the spool in particular history entry clicking small wrench icon.
What I need to add is table with summary info per spool (and/or chart)

Vikinge, 2017-02-10:
great.  I had seen the spool option in settings but no wrench.  I think the wrench might be hidden while printing, can you confirm?

imrahil, 2017-02-10:
yup. all buttons in my plugin are hidden during printing

Vikinge, 2017-02-10:
got it.  I have been working my printer pretty hard, printing most of the day.  Thanks .

closed 2017-03-09 by imrahil