imrahil/OctoPrint-PrintHistory #46: Plugin not working

motley74, 2017-02-24:
Sorry if this is not the best place for this but I have not found any other resources.  I installed your plugin on my new OctoPi installation and the plugin installed correctly but the history remains blank after a couple of days.  Is there some configuration I am missing to enable this?  I am able to connect to my printer and complete prints and I have a few other plugins installed that are working but I would like to get this one working as well.

Any help is appreciated, and thanks for creating this plugin.

madmax2, 2017-03-08:
same here

I added the filament and cost in the settings but nothing showing in history

al-jimenez, 2017-03-08:
I also have this problem.  Just installed it, restarted (cold start) and printed a successful print and one that I had to cancel, but I have not history.   There is no existing history.yaml file at all.  So it may never have gotten initialized.

imrahil, 2017-03-09:
Fixed in 1.2

closed 2017-03-09 by imrahil

eriecableguy, 2017-10-29:
Sorry, couldnt figure out how to start a new comment.
I have your DH for the Lightify dimmer switch. 
I have 2 on each side of my kitche , and sometimes they dim my smartbulbs, when all I want is the on/off functionality. How would I change the code, to remove all dimming features? 
TIA