imrahil/OctoPrint-PrintHistory #47: Filament usage not tracked every time

jgreth, 2017-02-26:
Hi I wonder under which circumstances (or what needs to be set) the used filement lenght gets tracked?
Sometimes there is a value - sometimes not. But used same slicer with same settings and both prints succeeded. 
Can you clarify? Or update the Readme / Docs?

That would be great.

See attaced image for what i mean. 
[image]

lexter12, 2017-06-10:
I got the same exact issue. Some prints (apparently randomly) don't show filament usage. 
[image]

scharrin2, 2017-10-20:
I have the same issue.

First I thought it was after installing the Filament Manager - but this issue was already before:
[image]

Best regards,
Christian