imrahil/OctoPrint-PrintHistory #48: printhistory failing with AttributeError: 'NoneType' object has no attribute 'parse'

pdvorak57, 2017-03-04:
I installed printhistory plugin on octoprint 1.3.1, and I don't have any print history information, and I am getting the following error in the logs:

2017-03-03 00:26:21,166 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2017-03-03 00:26:21,222 - octoprint.plugin - ERROR - Error while calling plugin printhistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugin/__init__.py", line 229, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/__init__.py", line 146, in on_event
    return eventHandler.eventHandler(self, event, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/eventHandler.py", line 46, in eventHandler
    parameters = gcode_parser.parse()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/parser.py", line 25, in parse
    parameters = self.parser_factory.parse(self.file)
AttributeError: 'NoneType' object has no attribute 'parse'

imrahil, 2017-03-04:
what sliced did you use for that gcode file?

pdvorak57, 2017-03-04:
cura 2.3.1 on Mac.  It fails for all the prints I am doing, do you need to see the gcode?

imrahil, 2017-03-04:
> do you need to see the gcode?

yes, please

pdvorak57, 2017-03-05:
[WDI_elf_ranger_supports.gcode.gz](https://github.com/imrahil/OctoPrint-PrintHistory/files/819390/WDI_elf_ranger_supports.gcode.gz)

danielei, 2017-03-05:
I have the same problem:

2017-03-04 07:41:29,925 - octoprint.plugin - ERROR - Error while calling plugin printhistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugin/__init__.py", line 229, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/__init__.py", line 146, in on_event
    return eventHandler.eventHandler(self, event, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/eventHandler.py", line 46, in eventHandler
    parameters = gcode_parser.parse()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/parser.py", line 25, in parse
    parameters = self.parser_factory.parse(self.file)
AttributeError: 'NoneType' object has no attribute 'parse'

danielei, 2017-03-05:
BTW - I have this problem when using Cura 15.02.1, when I use Simplyfi3d, it's ok.

danielei, 2017-03-05:
but before I installed Octoprint 1.3.1 it used to work on Cura as well.

armicron, 2017-03-05:
Parser searches for `Cura_SteamEngine` word on the first 3 lines.
Increasing this constant to 10 should help.
https://github.com/imrahil/OctoPrint-PrintHistory/blob/master/octoprint_printhistory/parser.py#L49

PR #49 (2017-03-05): Correct CURA detection
I've checked that it correctly detects [WDI_elf_ranger_supports.gcode.gz](https://github.com/imrahil/OctoPrint-PrintHistory/files/819390/WDI_elf_ranger_supports.gcode.gz)

Fixes #48 

p.s. Need to refactor parser, it shouldn't fail if nothing detected.
closed 2017-03-05 by imrahil