imrahil/OctoPrint-PrintHistory #54: Nothing Showing Up

pook0959, 2017-05-15:
Just installed this plugin fresh with a fully updated octoprint server. Nothing is showing up int he history tab. the tab is there with the framework, but it is not pulling in any print history. I installed via the plugin manager in octoprint. Any suggestions or missing steps that I can try?

zepco, 2017-06-20:
I have the same problem.

yo-k, 2017-10-19:
Same here. 2017-06-21-octopi-jessie-lite-0.14.0.img . octoprint updated to Version 1.3.5 (master branch) ota. Any suggestions?

jordisf, 2018-01-21:
Same here. Installed january 2018. Octoprint 1.3.6 / raspberry Pi 2 / OctoPi 0.14.0

toozinger, 2018-02-03:
It worked great for me until 2018. Last print was in December, and then January it stopped working.

toozinger, 2018-02-06:
I found a fix for my problem.  It is only tracking for NON Admin accounts!

yo-k, 2018-02-06:
> I found a fix for my problem. It is only tracking for NON Admin accounts!

...hmm. it works sometimes on admin accounts... and sometimes not. Does it work always on non admin accounts? Strange behavior.

yo-k, 2018-02-06:
Strange behavior  - maybe issue #72 ?

toozinger, 2018-02-07:
I don't have any folders (that I know of) so I don't believe so.

GaugeTheory, 2018-08-04:
Same problem here. Is there any solution?

alexisju, 2018-08-06:
Annoying issue indeed. Is this plugin still maintained??

toozinger, 2018-08-06:
I haven't had issues recently. My previous issues seem to have solved themselves. Sorry, I can't be of more help.