imrahil/OctoPrint-PrintHistory #60: Ability to use FilamentManager plugin

Mavyre, 2017-08-23:
Hello!  First of all, thank you for this great plugin!
The plugin FilamentManager just came out, and is very great. It provides an API to manage filament on OctoPrint. It'd be great if we could use it (and therefore display what print was printed with what filament automatically).

It's available here: https://github.com/malnvenshorn/OctoPrint-FilamentManager

Thanks!

keombre, 2018-05-22:
Hi, this would be really awesome! Also adding support for the CostEstimator plugin would allow us to see total print price. Maybe even cost per user. We have shared printer in school and deciding who's going to pay for the filament is always difficult.

Link to CostEstimator here: https://github.com/malnvenshorn/OctoPrint-CostEstimation

WildRikku, 2024-09-05:
There is also SpoolManager, which has more functionality than FilamentManager. Considering that this plugin, PrintHistory, is rather lightweight, I am not sure if people will use a heavy plugin like SpoolManager in combination with this one.

Also, both filament plugins, FilamentManager and SpoolManager, are currently abandoned.