imrahil/OctoPrint-PrintHistory #72: Files printed from folders are not in history

runam0K, 2018-01-08:
When I print a file from a folder, this file shows not in the history.

Octoprint Version 1.3.6
Octopi  Version 0.14.0

CKENT0369, 2018-01-12:
Same problem here.

jbkuma, 2018-01-26:
Does not work from folders at all from what I can tell.  I just assumed it stopped working until I saw this.

daemondazz, 2018-02-09:
I have also just installed this plugin and was not getting any print history. Moved the gcode file into the root of the folder tree and the print showed up in the history tab.

marcoslunag, 2018-04-12:
I have the same problem, any solution?

sandyeggoxj, 2018-04-27:
Same issue here. I assumed it wasn't working. I'll try to print from root and then move to appropriate folder to store it for future printing.

Airfoil73, 2018-06-24:
I have printed multiple items from g-code in the root and nothing.  I did pull in the g-code, I didn't slice it in OctoPrint.