imrahil/OctoPrint-PrintHistory #76: Feature Request: Allow spool tracking from the Filament Manager plugin

Burritobun, 2018-06-28:
The Filament Manager plugin is already a great plugin for tracking spool metrics. Why not hook into it for managing the spool inventory similar to how the Cost Estimation plugin hooks into the same plugin. All you'd have to do on the frontend is place a checkbox to allow users to use the Filament Manager plugin instead of rewriting their entire spool inventory.

masterki, 2018-07-11:
Great idea! Pleas make this possible!!!

SirBastol, 2019-01-02:
I second and third this.. 
if detect spool manager plugin  let it override the details  costs and such
ALSO failed prints should recaclulate used filament based on when it was cancelled..  even if its a simple time versus max amount maths..

svalavuo, 2019-02-13:
Would like too to see this happen.

r-a-i, 2019-02-20:
Yes! Please!

mwmason, 2021-12-31:
Just re-upping this feature request - as a newbie it was confusing that it wasn't already integrated with Filament manager [my forums post](https://community.octoprint.org/t/filament-manager-vs-filament-manager/40196/2)

guysoft, 2022-01-03:
It not something I have the time to add, if you have a PR for this will gladly review test and merge

WildRikku, 2024-09-02:
Related #60 

I think this would be awesome, too, and I might have time to implement this. I would suggest the following:

If filament manager is present, offer both filament manager spools and print history spools in the edit dialogue. Automatically set the spool to the one selected in filament manager. Show a hint in the print history spool settings that it is not necessary to setup spools when filament manager is activated.

This should leave print history usable without filament manager and cause only few problems when filament manager is uninstalled somewhen. I am not (yet) sure what to do with spools from filament manager logged in print history when filament manager is uninstalled.