imrahil/OctoPrint-PrintHistory #77: Plugin not working

GaugeTheory, 2018-08-05:
Hello all

I installed plugin history but sometimes it works, sometimes it doesn't. I have printed more than 10 jobs, but I only have 2 entry on history. On log file I have that error.

2018-08-05 14:24:13,248 - octoprint.plugin - ERROR - Error while calling plugin printhistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 225, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/__init__.py", line 146, in on_event
    return eventHandler.eventHandler(self, event, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/eventHandler.py", line 46, in eventHandler
    parameters = gcode_parser.parse()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/parser.py", line 28, in parse
    parameters = self.parser_factory.parse(self.file)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/parser.py", line 41, in parse
    parameters.update(self.parse_bottom(gcode_file))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/parser.py", line 90, in parse_bottom
    config.readfp(io.BytesIO(settings))
  File "/usr/lib/python2.7/ConfigParser.py", line 324, in readfp
    self._read(fp, filename)
  File "/usr/lib/python2.7/ConfigParser.py", line 512, in _read
    raise MissingSectionHeaderError(fpname, lineno, line)
MissingSectionHeaderError: File contains no section headers.
file: <???>, line: 1
'{"extruder_quality": ["[general]\n'

NewbieBFU, 2018-09-23:
yes, I experience the same problem. It seems that everything I slice with Slic3r is recorded to history and things I slice with Cura are missing :(

petrows, 2018-11-11:
Have the same problem, don't see any jobs in list since install. Using Cura 3.5.1

KostteT, 2018-11-24:
+1 doesn't work (cura 3.6 )

smnvhn, 2018-12-18:
Have same issue

luisromano91, 2019-03-14:
I'm not sure why, but after over a year and 14 pages of printed parts, the history plugin just stopped counting the new ones I print. The last count is from 10 days ago, but I haven't had a single day without printing ever since and none of the new parts appear on the list. I've only been using Slic3r PE with the same configurations, so I assume the problem is not the gcode. Also there was no modification on the octo or any other feature, aside from a 15% reduction on the print feedrate of the parts I've been dealing with to avoid wiggle. Can anybody give me a clue on what can be done?

guysoft, 2021-07-27:
@luisromano91 Fixed in last #89

closed 2021-07-27 by guysoft