imrahil/OctoPrint-PrintHistory #83: Missing filament usage

kb-, 2019-03-11:
The filament usage sometimes doesn't show up in the history despite being tracked properly by Octoprint.

PrintHistory 1.2
OctoPrint 1.3.10
OctoPi 0.15.1, running on Raspberry Pi 3 Model B Plus Rev 1.3

[image]

gif3, 2019-03-23:
For me it happens a lot also.

It looks like, if I choose model to print and quickly click print, then the filament usage is missing.
If I choose model, then wait a second or two, then click print, then the filament usage is recorded.

When I printed directly from Slic3r PE with "Send to printer" (use Octoprint API) and when checked "Start printing after upload", then the filament usage is always missing.

(OctoPrint: 1.3.10 on Raspberry Pi 3, Print History Plugin: 1.2, Slic3r PE 1.41.3+win64)

jolcese, 2019-05-09:
I fixed this bug here: https://github.com/jolcese/OctoPrint-PrintHistory/commit/1887145a66ad532f46eeceac716beb5be3a13f10

gif3, 2019-09-07:
Thank you very much.

PS: At first I didn't know how to install, so for others: choose install "from url" and use "https://github.com/jolcese/OctoPrint-PrintHistory/archive/master.zip"

guysoft, 2021-07-27:
Fixed by #89

closed 2021-07-27 by guysoft

haarp, 2022-02-08:
@guysoft  I'm still seeing empty filament usage in version 1.3 of this plugin. I'm using Prusaslicer's Upload&Print button, if that matters. Could it be related to #27, i.e. Octoprint not having parsed filament usage yet before the print starts?

guysoft, 2022-02-08:
Should be the latest.
There are 4 commits in the master branch that might solve the issue.

If you can test and see if it solves the problem it would be great!

1. Run
```
~/oprint/bin/pip install git+https://github.com/imrahil/OctoPrint-PrintHistory@devel --upgrade
```

2. 
restart octoprint

3. test

If this solves the issue I will release 1.4

haarp, 2022-02-09:
@guysoft 

Thanks. The supplied command didn't seem to upgrade the plugin, so I manually edited `~/oprint/lib/python3.7/site-packages/octoprint_printhistory/export.py` with the newest `devel` branch version and restarted Octoprint. Unfortunately the problem remains.

It only seems to happen when doing Upload+Print from Prusaslicer, which presumably doesn't give it time to parse the metadata, similar to #27. If I upload manually, and then press print, the filament usage is correctly entered into the history.

guysoft, 2022-02-10:
How did you install the plugin? 
That command only works on OctoPi.

What kind of error are you getting now in the log? It might not be the tabs issue from before.