imrahil/OctoPrint-PrintHistory #86: History stopped getting populated

kb-, 2020-04-27:
I updated Octoprint to the version 1.4.0 and noticed the history stopped getting populated.
(OctoPi 0.15.1, running on Raspberry Pi 3 Model B Plus Rev 1.3)

stidf, 2020-04-27:
Just happened to me as well. Stopped working about 2 weeks ago on April,
12th according to the log.

anhtuan445, 2020-04-28:
Check this topic:  https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/23

You should try to find the same error in this plugin source files and replace it with name/path as appropriate. Let us know if it works ;)

anhtuan445, 2020-04-28:
For anyone looking for a quick fix, this is how I solved it for now:

Change the file for your plugin - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory/eventHandler.py

1. Change payload["file"] to payload["path"] (look for all places in this file and replace it)
  fileData = self._file_manager.get_metadata(payload["origin"], payload["path"])
2. Add your upload directory to this line like so:
  UniversalParser('/home/pi/.octoprint/uploads/'+payload["path"], logger=self._logger)
3. After saving be sure to restart OctoPrint

Now my prints are added to history again :)

kb-, 2020-04-28:
Thank you! It wasn't working with payload["path"] either. But changing payload["file"] to payload["name"] as mentioned in the topic you mentioned does the job!

anhtuan445, 2020-04-29:
Thanks, corrected my post and added more detail.

guysoft, 2020-08-02:
Can I close this? Seems you solved the issue.
Also, should we add @anhtuan445 solution to FAQ/wiki?

guysoft, 2020-08-07:
Ah wait, you solved it by editing the code @anhtuan445 Is this worth a pull request?

anhtuan445, 2020-08-07:
Sorry don't understand what you mean.

I just edited some files to get it working for me and put the info here so others can also benefit from it.

guysoft, 2021-07-27:
Fixed by #89

closed 2021-07-27 by guysoft