imrahil/OctoPrint-PrintHistory #90: Switch default branch to dev and keep master as release branch

guysoft, 2021-07-28:
Hey @imrahil 

Happy to update the plugin is back up and working.
I want to change the default branch to "devel" and use the "master" branch each time we push a new release.

Could you either give me permissions to change the default branch, or alternatively, change it yourself to devel?

Thanks!

imrahil, 2021-07-28:
Changed. Is it ok now?

guysoft, 2021-07-28:
Nope, i have no access to settings.
I think you should just change the default branch, after reading i am not sure its possible for a user repo to give access to this. I think that is why they created organizations and such.

imrahil, 2021-07-28:
I couldn't find the way how upgrade your privileges so I just changed the default branch... but I don't have time to test it 😁

guysoft, 2021-07-28:
I think its best if you just change the default branch, its under "Settings>Branches' then there is a default branch dropdown, set it to devel and not master