imrahil/OctoPrint-PrintHistory #96: Feature Request: Automatically Log which user started print

jmiltxtav, 2023-05-31:
I have an instance where around 15 people have access to the printer. I want to log which user starts the print. 

I see the option to go in afterward and mark which user started the print, but I would like that to be automatic because they will not go back and edit the history. From what I see in the code, we might be able to add a column using the [printing event](https://docs.octoprint.org/en/maintenance/events/index.html#printing) hook. 

I can attempt to add this as well, but I am unsure how everything relates to everything else, so I would prefer if one of the devs could add it.

guysoft, 2023-11-07:
You are welcome to try and add this. I have my hands full at the moment and would not get to this in the future

WildRikku, 2024-09-02:
this is also kind of a duplicate of #20 (even though that issue originally was about being able to log the user at all, now it's about automatically adding the user, which seems not to be possible currently)

PR #103 (2024-09-03): Use 'owner' payload from PRINT_DONE which is now available. Fixes #20…
… and also fixes #96.

See https://docs.octoprint.org/en/master/events/index.html#printing - it listed "owner" for the payload so I just used that. I guess it was added in OctoPrint 1.4 and this plugin is older than that.
commit c6e85a7f (2024-09-03): Use 'owner' payload from PRINT_DONE which is now available. Fixes #20 and also fixes #96.
closed 2024-09-03 by guysoft