imrahil/OctoPrint-PrintHistory #98: No thumbnail in history list if spaces/accents in filename

PyBonnetainNesterenko, 2023-06-05:
Hello,
The plugin works great, but the thumbnails are not displayed if the printed filename contains (at least) spaces and accents. Which is annoying.

Fist with accents :
[image]

A proper display and print :
[image]

Renamed the previous file to add some spaces (and printed/canceled it, it's only to show the problem) :
[image]

guysoft, 2023-06-09:
Wrong issue tracker, you want the OctoPrint-PrintJobHistory plugin, this is the OctoPrint-PrintHistory plugin.

https://github.com/OllisGit/OctoPrint-PrintJobHistory

This one sadly has no picture support.

closed 2023-06-09 by guysoft

PyBonnetainNesterenko, 2023-06-10:
Sorry for the mistake. Must be old age setting in.

guysoft, 2023-06-10:
I hope not, happend to me too a few years back😅
https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/68#issuecomment-667654896